BluEye-Robotics/blueye.sdk #83: Add schema definitions for HTTP API

follesoe, 2020-04-20:
The `http-api.yml` definition references external schema definitions for custom JSON response types. These schemes should be defined and included in the SDK repository, to make the HTTP documentation complete.

I.e. `$ref: '#/components/schemas/DroneInfo'`, need to actually include the `DroneInfo` schema. Relevant for all HTTP end-points returning JSON documents.

sindrehan, 2020-04-20:
Not sure I understand what you mean, aren't they included at the end of document? eg. https://github.com/BluEye-Robotics/blueye.sdk/blob/e508673ab5b2d8c00dc3f1bba47eeb5b0231d13e/http-api.yml#L633

Or do you  want them defined some other way?

follesoe, 2020-04-21:
My bad @sindrehan, I saw the schema for request part of the end-points but not the schemas, but they are there. Thanks. Closing the issue. The thinking was that if we have a complete schema definition we can generate clients for statically typed languages like C#, which we now can as the schema definitions actually are in place 👍

closed 2020-04-21 by follesoe