BluEye-Robotics/blueye.sdk #88: Link not working in quick_start.md

johannesschrimpf, 2020-10-01:
**Describe the bug**
Link to "Reference section" does not work on https://github.com/BluEye-Robotics/blueye.sdk/blob/master/docs/quick_start.md

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://github.com/BluEye-Robotics/blueye.sdk/blob/master/docs/quick_start.md
2. Click on "Reference section"
3. See error

**Expected behavior**
A page is shown

johannesschrimpf, 2020-10-01:
It seems to work here:
https://blueye-robotics.github.io/blueye.sdk/docs/quick_start/

sindrehan, 2020-10-01:
The documentation is not really configured for being browsed directly from the markdown files. And to support browsing the [documentation locally](https://blueye-robotics.github.io/blueye.sdk/docs/quick_start/#local-documentation) I can't link directly to the generated docs either.

closed 2022-01-02 by sindrehan