BluEye-Robotics/blueye.sdk #124: Update or remove GIFs

sindrehan, 2023-03-10:
Currently the GIFs being used around the documentation are out of date (eg. the on in the README uses `Pioneer()` instead of `Drone()`). This can be confusing for the users, so either we need to take the effort of updating them, or remove them entirely.

johannesschrimpf, 2023-03-10:
I think the gifs are very nice and descriptive and that we should put in the effort updating them.

follesoe, 2023-03-17:
Mathilde and Petter can also help create GIFs or other marketing content. We could also use something else than GIFs. I know GitHub has a "type out code sync with video" type of animation thing, if we want to sync code line-by-line with action on video.

But I agree, it is a cool way to show the simplicity of the remote Python SDK 👍

closed 2026-10-28 by sindrehan