BluEye-Robotics/blueye.sdk #183: Blueye.sdk buffer for motion commands?

Urlaxle, 2025-02-13:
Hi,

when attempting to control the Blueye through the blueye.sdk establishing a connection and running commands as you do in the github demonstration works great, it also seems to work great if we send commands spaced out. However, a delay builds up if the motion commands are sent at high frequencies. 

As an example today I connected to the Blueye with the intention of tuning a PID controller for the different DOFs. I initialized my control system, including the control loop running at 100Hz (at this point just sending sway, surge, heave and yaw values to 0.0 over and over), and spent some time making sure that all the sensors and state estimations were working properly. When it became time to test the PID in the heave direction the drone didn't move even though the *drone.motion.heave=value*  clearly was set to non-zero values. 

I attempted to set different parameter values to increase the output of the PID, but nothing seemed to make the thrusters execute. Then I shut-off my system and went into the BlueyeApp.  To my suprise the drone did not react to the manual commands given inside the app either. I logged out of the BlueyeApp and went outside to turn the drone off-and-on, but then suddenly it started to dive to the bottom. In the end I had to drag it up by the tether and turn it off in the water as it continued to dive for a long time. 

I am wondering if this behavior can be explained by a buffer in the Blueye queuing up the commands much faster than its able to execute them? Tomorrow I will try to heavily reduce the frequency of the controller to see if it helps.

Urlaxle, 2025-02-14:
Attempted to run my control loop today with 10Hz, and it works fine without building up any delay.

aviggen, 2025-02-17:
Ok, The app publishes messages at a fixed `20 Hz` update rate, so it could be that the drone simply cannot keep up when going up to 100 Hz. You can try with `20 Hz`. But there is not much we can do now to allow `100 Hz` without major rework.

closed 2025-02-17 by aviggen