BluEye-Robotics/blueye.sdk #184: Support for Setting Custom GPS Start Location in Missions via SDK

AbubakarAliyuBadawi, 2025-02-20:
Hi,

I have been working with the blueye mission planner for quite some time now and would like to know if I can set a specific GPS coordinate to start a mission rather than using the drone's location. I know it is possible form the app but would like to know if I can set that using the sdk in a python script.

Thanks

aviggen, 2025-02-21:
If I understand your question correctly, you would like to reset the drone position to a `manual coordinate`, which is equivalent to the reset to `POI` in the app?

It does not seem like the `Reset Position` has been added to the SDK yet, but I can add that as a feature request.

If you would like to try it out yourself, you need to populate the [ResetPosition](https://github.com/BluEye-Robotics/ProtocolDefinitions/blob/2546aed58af3cfe4cbe502bac0af2ed53f978969/protobuf_definitions/message_formats.proto#L419) message and send the [ResetPositionCtrl](https://github.com/BluEye-Robotics/ProtocolDefinitions/blob/2546aed58af3cfe4cbe502bac0af2ed53f978969/protobuf_definitions/control.proto#L119).

You can probably extend the [CtrlClient in connection.py](https://github.com/BluEye-Robotics/blueye.sdk/blob/953767380bd0e1cdd865749a3c4d90d876c07f6a/blueye/sdk/connection.py#L394).

If you actually want to start a mission when the drone is within a POI (2 m circle of acceptance), we do not have this functionality, but you could make that logic in your Python Mission Planner. If within COA, run_mission.

AbubakarAliyuBadawi, 2025-02-24:
Exactly, I want to reset the position to a manual coordinate, as almost every time the mission gets aborted due to the drone's position being unknown. Thanks, I'll give it a try before the feature is implemented.

Another error we often encounter is when a waypoint mission is set with the mission planner: when the mission is running, we usually get an error saying "Drone is stuck, please check the tether!" and then the mission gets aborted automatically, despite the tether being OK.

We want to use the planner on a field trip this week and wanted to make sure we can trust it.

Thanks.

[image]

aviggen, 2025-02-24:
I see. The message you get corresponds to the notification, `NOTIFICATION_TYPE_DRONE_IS_STUCK`, which is published when the drone velocity gets below a certain level. These are individual checks for surge and heave depending on if it is a `go to depth` or `go to WP` instruction. There is currently no way for you to change this threshold.

But you can always resume the current mission if you hit `drone.mission.run()`.

AbubakarAliyuBadawi, 2025-02-24:
Ohh I see...

Thanks for your prompt response, how about when using the mission planner app, do I need to manually run the mission again or I can modify the drone velocity since it is not possible from the SDK, right now we want to make use of the planner app for the first phase of the project, If everything works fine we can transition to use the sdk.

Thanks

aviggen, 2025-02-24:
It's not yet possible to change the default surge reference speed from the app, so it is always `0,6 m/s` if you have a DVL connected. If it's using other position sources, it's just a constant thrust force independent of speed.

To resume an aborted mission in the app you simply hit "Play" on the map again.

AbubakarAliyuBadawi, 2025-02-24:
Alright, thanks.

closed 2025-05-12 by aviggen