BluEye-Robotics/blueye.sdk #189: Blueye File Transfer on Ubuntu does not show Multibeam videos

leoguen, 2025-05-03:
Hi Blueye Team, 

it seems like the Blueye File Transfer on Ubuntu does not support Multibeam files. I cannot see them here in my ubuntu GUI but on my phone (downloaded them that way now). 

[image]

follesoe, 2025-05-04:
Thanks for your request @leoguen. 

The multibeam recordings are not actual video files, but rather log files containing a series of `BinlogRecord` messages, wrapping `MultibeamPingTel`, which in turn contains a [MultibeamPing message](https://blueye-robotics.github.io/blueye.sdk/latest/reference/blueye/protocol/message_formats/#blueye.protocol.types.message_formats.MultibeamPing).

So to view a multibeam recording you have to play it back in the Blueye App on iOS, macOS or Android. From the Blueye App it is possible to export the multibeam recording as a `.mp4` video file for sharing with users who do not have the Blueye App, or for playback on platforms where the Blueye App is not available (Windows/Ubuntu).

We have not prioritized adding support for downloading `.mbez` (the extension we use for multibeam recordings, short for Multibeam Blueye Zipped) in the File Transfer App. We will probably not add this neither.

The files are still accessible if you want to download them and write code to process them using the Blueye SDK (which supports reading log files). I also have some information about the Blueye log file formats covered in a presentation from NDC Tech Town 2024: https://youtu.be/LJCQA4UhS58?si=EBIdTSpiySrcYoOz&t=2907

The files can be downloaded over the WebDAV (HTTP) protocol at http://192.168.1.101:5050. You can even use third-party file transfer applications that support WebDAV to connect to this end-point to manage media files.

We will probably not offer video export directly from the drone no the current hardware platform, as the conversion time would very long on the X3 compute module.

I hope this answers your questions, if so, feel free to close the issue.

follesoe, 2025-05-04:
One additional note: users do not need access to a Blueye drone to play back a `.mbez` file, so if you have a mbez file it can be transferred to an iOS, macOS or Android device and they can use the Blueye App (without being connected to a drone) to open the file. So think of the Blueye App as your player for this file format.

leoguen, 2025-05-07:
Great, thank you for this explanation!

closed 2025-05-07 by leoguen