BluEye-Robotics/blueye.sdk #190: Mission Abort in Mission Planner

leoguen, 2025-05-03:
Hi Blueye Team, 

during our recent campaign we tried to use the Blueye Missionplanner to maneuver to a POI only 10-20m away but we kept on getting mission abort messages without any other information. Is there any way to display more abort information for debugging?

Cheers, 

Leo

follesoe, 2025-05-05:
Hi @leoguen,

@aviggen is out of office until Wednesday, so he will not be able to provide feedback to the issue until towards the end of the week.

Just giving you a heads up that it will take some days until someone will reply.

aviggen, 2025-05-08:
Hi Leo. Try to have a look at the [NotificationTel](https://github.com/BluEye-Robotics/ProtocolDefinitions/blob/7966a959f0c09d2447d956d22a99220fe3959d2a/protobuf_definitions/message_formats.proto#L484) message. You'll first see the `NOTIFICATION_TYPE_MISSION_ABORTED` message followed by one of the messages below:
- `NOTIFICATION_TYPE_DRONE_POSITION_IS_UNKNOWN`
- `NOTIFICATION_TYPE_WAYPOINT_IS_TOO_FAR_AWAY`
- `NOTIFICATION_TYPE_DRONE_IS_STUCK`

You can use the Foxglove application, [as described here](https://blueye-robotics.github.io/blueye.sdk/latest/foxglove-bridge/), to subscribe to the notification messages. If you use the State Transition view, you can easily read the Notification enums on a timeline. 

You can also check if the `is_valid` field on the PositionEstimateTel is valid. When this goes invalid, it will abort the mission.

Just let me know if you need help with foxglove and Docker. Hope this will give you some more information on what is going on!

aviggen, 2025-05-08:
This is what it will look like in Foxglove when a mission is aborted due to an invalid position estimate. @leoguen 

[image]

closed 2025-06-05 by sindrehan