BluEye-Robotics/blueye.sdk #191: Turning Blueye on and off remotely

leoguen, 2025-05-06:
Hi Blueye Team, 

for our resident Blueye project we are considering whether we can turn the drone on and off remotely to save battery (or maybe standby state) is there a way to do this through the tether/GP or can you only turn the drone on/off with the magnet?

Thanks, 

Leo

sindrehan, 2025-05-07:
Hi Leo,
Unfortunately there is no way to turn the drone on/off remotely without using a magnet. In theory it's possible to shutdown the main computer (just using the regular linux `shutdown` command), but you would have no way to turn it back on again.

follesoe, 2025-05-07:
If the drone is not using the thrusters or lights, it consumes very little power. You can listen to the battery telemetry messages and read the fields:

- `average_time_to_empty`
- `average_time_to_full`

When docked with the inductive charger to get an estimate on how much runtime you have left/until full if charging.

You can also check idle power using `average_current` (don't remember the window size for this average but belive it is 30s?).

As @sindrehan said turning it on/off through external mechanisms is not supported. We do not have this feature on our roadmap, but its something that has been asked for occasionally for USV or ROV-flyout scenarios.

leoguen, 2025-05-13:
Great thank you for the information!