BluEye-Robotics/blueye.sdk #192: Plot GPS position and mission planner reference using the SDK

AbubakarAliyuBadawi, 2025-05-19:
[image]

Hi Blueye Team,

I was wondering if it is possible to visualize the lat and log of the drone using Foxglove, not as a raw message, but as lat and log, so basically plotting (blueye.Protocol.LatLongPosition). I tried using the "Map" panel, but could not visualize it.

Another question is if it's possible to get the reference position of the drone when I set up multiple missions using the mission planner. I want to create a plot of the estimated and desired position of the drone in north, east, heading, and depth during the mission.

follesoe, 2025-05-19:
For your first question, this is possible if you transform the Blueye message into a `LocationFix` message, which is part of the known schemas of Foxglove.

Here is an example I wrote about a year ago that would plot the pilot position, but the same would be the case for the position estimate message:

```ts
*Foxglove script to transform a PilotGPSPositionTel*

```// The ./types module provides helper types for your Input events and messages.
import { Input } from "./types";
import { LocationFix, PositionCovarianceType } from "@foxglove/schemas";

// These are the topics your script "subscribes" to. Studio will invoke your script function
// when any message is received on one of these topics.
export const inputs = ["blueye.protocol.PilotGPSPositionTel"];

// Any output your script produces is "published" to this topic. Published messages are only visible within Studio, not to your original data source.
export const output = "/studio_script/PilotGPSPositionTel";

// This function is called with messages from your input topics.
// The first argument is an event with the topic, receive time, and message.
// Use the `Input<...>` helper to get the correct event type for your input topic messages.
export default function script(
  event: Input<"blueye.protocol.PilotGPSPositionTel">,
): LocationFix {
  return {
    frame_id: "0",
    timestamp: {
      sec: 0,
      nsec: 0,
    },
    altitude: 10,
    latitude: event.message.position.latitude,
    longitude: event.message.position.longitude,
    position_covariance: [0, 0, 0, 0, 0, 0, 0, 0, 0],
    position_covariance_type: PositionCovarianceType.UNKNOWN,
  };
}```

follesoe, 2025-05-19:
And this one uses the position estimate to produce a geo JSON string:

```ts
// The ./types module provides helper types for your Input events and messages.
import { Input } from "./types";
import { GeoJSON, PositionCovarianceType } from "@foxglove/schemas";

// These are the topics your script "subscribes" to. Studio will invoke your script function
// when any message is received on one of these topics.
export const inputs = ["blueye.protocol.PositionEstimateTel"];

// Any output your script produces is "published" to this topic. Published messages are only visible within Studio, not to your original data source.
export const output = "/studio_script/PositionEstimateTel";

// This function is called with messages from your input topics.
// The first argument is an event with the topic, receive time, and message.
// Use the `Input<...>` helper to get the correct event type for your input topic messages.

var path: Array<Array<number>> = [];

export default function script(
  event: Input<"blueye.protocol.PositionEstimateTel">,
): GeoJSON {
  var longitude = event.message.position_estimate.global_position.longitude;
  var latitude = event.message.position_estimate.global_position.latitude;

  
  path.push([longitude, latitude]);
  if (path.length > 100) {
    path.shift()
  }

  var geojsonData = {
    type: "FeatureCollection",
    features: [
      {
        type: "Feature",
        geometry: {
          type: "LineString",
          coordinates: path,
        },
      },
    ],
  };

  return { geojson: JSON.stringify(geojsonData) };
}

// https://w7.pngwing.com/pngs/247/15/png-transparent-arrow-arrow-angle-triangle-black.png
```

follesoe, 2025-05-19:
You can find more information about message converters here: https://docs.foxglove.dev/docs/visualization/extensions/guides/create-message-converter

AbubakarAliyuBadawi, 2025-05-20:
Thanks, that would help a lot.

closed 2025-05-20 by AbubakarAliyuBadawi