BluEye-Robotics/blueye.sdk #193: Long Term Deployment Learnings

leoguen, 2025-05-20:
Hi Blueye Team,

We’re preparing for a two-week continuous deployment of the Blueye, operating at depth and charging via the inductive tether connected to a subsea rig.

I wanted to ask if you have any insights, best practices, or known issues related to such long-term deployments. For example, we heard that earlier there may have been issues with log file corruption when the system was running for extended periods—are there similar issues or are there workarounds we should be aware of?

Have you or other users conducted deployments of similar duration? Any advice or lessons learned would be highly appreciated.

Thanks in advance!

Best regards,
Leo

follesoe, 2025-05-20:
Hi @leoguen,

By default, the Blueye drone will turn off logging of dive log files after 6 hours. This is to prevent cases where customers forget to turn their drone off, and is connected to wall power, so the drone would run indefinitely and fill the data partition.

We do not have an SDK/API end-point to change this behavior. You can change publishing frequency, but not logging frequency, for topics, and we cannot control the auto-off logging.

So if you need logging you would have to use the onboard SDK and log the data you want, how you want it.

No user has conducted long-term deployments like that. We have had drones powered on for > 24h on the high capacity battery with out issues. We have also had bench-drones (our development kits) powered on for weeks without memory leaks or  software crashes.

So as long as it remains waterproof, you provide constant power, it should work except for the logging.

leoguen, 2025-05-20:
Ok, thanks for the feedback! Let me know if something comes to mind, I will keep you up-to-date if we run into some new issues!

closed 2025-05-20 by leoguen