BluEye-Robotics/blueye.sdk #195: Question about the fields in the PositionEstimate telemetry

Urlaxle, 2025-05-25:
Hi,

we were attempting to project pings from a sonar onto a map using the PositionEstimate message found in the documentation, https://blueye-robotics.github.io/blueye.sdk/v2.3/protobuf-protocol/#blueye-protocol-Attitude. However, we noticed that the sonar pings did not project into the scene where we would expect them based on the vehicles movement. 

Our assumption was that northing would be position in meters north and easting would be the position in meters east from the initial or reset point while heading would be with regards to north. To sanity check our assumption we ran some tests where we turned the vehicle to 0, 90, 180 or -90 degrees and tried to do straight translations. With our assumption we would expect the corresponding position either x or y based on heading to increase while the other should stay the same. This was not the case, and we also noticed that same movement gave different changes in x and y after rebooting the vehicle. See drawing below. 

[image]

This made us think that the x and y coordinate of the used navigational frame is based on the heading of the vehicle at boot similar to the waterlinked DVLs dead reckoning instead of using north-east convention, and the changes in x-y in our tests made a lot more sense when looking at the heading displayed at the DVL page 192.168.1.99 than the heading displayed in the PositionEstimate message. 

As we want to process the sonar pings we need to know the heading in the frame used for navigation rather than the north-east frame which the PositionEstimate seems to give us which makes our projection weird. I was wondering if there is a way to get this heading with the current telemetry? The Attitude message yields the same heading as the PositionEstimate. If not do you think it would make sense for us to use the heading given by the dead-reckogning report given by the DVL?

aviggen, 2025-05-27:
What you are doing should be correct. The `northing` and `easting` field of the `PositionEstimate` msg should be meters North and East from the starting point (NED). Then it depends on you ResetSettings in the navigation page, (settings tab) if you have set a manual start heading, or if you use the drone's compass. The entire track is rotated by this heading, so it's crucial to get it right. Make sure the compass heading is correct (i.e. not in a magnetic area) when you hit reset, if not, set a manual heading (that you can take out from a map). You should always use the `heading` field in the PositionEstimate message when working with the Northing Easting. The `yaw` in attitude might differ as it is representing the compass readings, (while the PositionEstimate.heading might be gyro only).

You can also try switching between DVL and Compass as heading source, to see what works best for you. The DVL option might be easier to start with as it is not affected by magnetic fields along the way. Make sure to calibrate it. Best practice is to place it in the water and wait until it's temperature has converged to the operating temperature, then take it out, and place it on solid ground. The calibration is persistent, so should not be necessary to do this prior to every dive. 

When it comes to the sonar image coordinates, they will be in body frame. So you would have to rotate this by the PositionEstimate.heading. 

And yes, the position reported in the WL DVL protocol is relative to the start heading, and not North.

Hope this helps, if not, just shoot again!

Urlaxle, 2025-05-30:
We performed the calibration you suggested before launching the Blueye on Wednesday and then it seemed to use north as its navigational frame. Since it was working we prioritized our testing instead of trying to recreate the issue, but I will attempt to run a more thoroughly test with multiple reboots when I get the time in the coming days. I will let you know how it goes and close the issue if I don't get the same behavior again. 

Thank you for your help.

follesoe, 2025-05-30:
@Urlaxle Great. We will close the issue for now. If you reproduce the error or have any additional issues feel free to re-open it.

closed 2025-05-30 by follesoe