BluEye-Robotics/blueye.sdk #198: Accessing the tilting feature of Sonar Servo

Sam1-2, 2025-05-31:
I am currently working with the Blueye at TBS with the sonar mounted at the bottom and connected to a servo for tilting. I was wondering if there is a way to access the servo to achieve the tilt and get relevant high frequency angle data using the sdk?

follesoe, 2025-05-31:
@Sam1-2 Yes; the Multibeam Servo is controlled by the `MultibeamServoCtrl` message, which holds a `MultibeamServo` message where you can set the tilt angle (-30..30).

To read the tilt angle you can subscribe to the `MultibeamServoTel` message.

If you look in the source code of the `CtrlClient` you can see how the various `Ctrl` messages are produced and published to the drone, such as the `set_lights` method: https://github.com/BluEye-Robotics/blueye.sdk/blob/master/blueye/sdk/connection.py#L233

I see that we have not exposed a method such as `set_multibeam_tilt` on the `CtrlClient`, so you would have to clone the SDK repo and extend it yourself. A PR would be appreciated. @sindrehan might be able to provide some example code showing how to publish an arbitrary `Ctrl` message by getting the `CtrlClient` out of the `Drone` object?

For receiving the tilt angle, you must subscribe to the `MultibeamServoTel` message. In the SDK there is an example script called `print_imu` that demonstrates how to subscribe to arbitrary messages (that are not exposed as properties on the `Drone` object`): https://github.com/BluEye-Robotics/blueye.sdk/blob/master/examples/print_imu.py

Sam1-2, 2025-05-31:
Thats great! I will implement it and update here soon. Thanks!

PR #202 (2025-06-05): Add servo control
Closes #131 
Closes #198