BluEye-Robotics/blueye.sdk #205: Protobuf IMU messages

Urlaxle, 2025-06-16:
Hi,

during the summer we are planning to run some experiments with the Blueye where the IMU data is going to be essential. From the protobuf documentation I can see that there are three separate IMU topics, raw data for IMU 1 and 2, and the calibrated IMU telemetry, which all provides IMU data in the body-frame of the vehicle. 

In order to decide which message to subscribe to we are wondering what the calibrated IMU messages entails. Is it bias corrected IMU measurements, or it something else? 

Another thing we are wondering about is if you have an approximate lever-arm estimation from the body frame these measurements are given in to the waterlinked dvl when it is mounted to the Blueye with your standard dvl mount. 

I am sorry if any of this is stated somewhere in the documentation, but I have not been able to find it. 

Best regards,
Ambjørn

follesoe, 2025-06-17:
If this answers your question @Urlaxle feel free to close the issue, or just let us know if you have additional questions 👍

Urlaxle, 2025-06-18:
I see, so does that mean that the IMU data sent to the subscriber is simply the most recent single raw/calibrated IMU measurement you have available? 

Sorry to go on about these details, but it is important that we are certain what the measurements are for the work we are planning.

aviggen, 2025-06-20:
Yes, that is correct, it will sample the most frequent value at 100 Hz. The internal loop is approximately twice as fast. 

To get higher rates it is possible to use the internal SDK to get access to the _libguestport_ IMU driver. But then you cannot run the normal control system alongside with it, which complicates things.

What are you planning to do?